bnicenboim/eeguana #2: Maybe an issue with summarise_all + t.test

auskate, 2018-10-17:
I get the error "Evaluation error: data are essentially constant" when I do this:

dat_stat <- dat %>%
      group_by(sample) %>% 
  summarise_all(tvalue = t.test(.[cond=="b"], .[cond=="d"])$statistic)

Apparently it's suggesting that the t.test is evaluating inappropriate data types (?), possibly because summarise_all summarises everything except the arguments I give it? Although it works in your vignette.

bnicenboim, 2018-10-18:
you tried with `.sample_id`, right? that's the correct column name

auskate, 2018-10-19:
In a previous version I tried with both, same error. I just tried it in the
current version with .sample_id and I get a new error: not enough X
observations.

bnicenboim, 2018-10-19:
try it again now, if it still doesn't work, please link some data where it fails

auskate, 2018-10-20:
I put data from 2 subjects here: https://www.dropbox.com/s/w4dm521dyv86n9i/ttest_test.rds?dl=0

It should be enough observations, but I get the same error even if I use the whole dataset. Traceback is attached, in case it's useful.

[image]

bnicenboim, 2018-11-13:
while it works in the vignette, this is not working with a minimal example:

```
data <- eeg_lst(
  signal = signal_tbl(
    signal_matrix = as.matrix(
      data.frame(X = sin(1:20), Y = cos(1:20))
    ),
    ids = rep(c(1L, 2L), each = 10),
    sample_ids = sample_int(rep(seq(-4L, 5L), times = 2), sampling_rate = 500),
    dplyr::tibble(
      .name = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
      radius = NA, .x = c(1, 1), .y = NA_real_, .z = NA_real_
    )
  ),
  events = dplyr::tribble(
    ~.id, ~type, ~description, ~.sample_0, ~.size, ~.channel,
    1L, "New Segment", NA_character_, -4L, 1L, NA,
    1L, "Bad", NA_character_, -2L, 3L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 2L, "X",
    2L, "New Segment", NA_character_, -4L, 1L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 1L, "Y"
  ),
  segments = dplyr::tibble(.id = c(1L, 2L), recording = "recording1", segment = c(1L, 2L))
)
```

the first works, the other two don't!
```
data %>% group_by(.sample_id) %>%  summarize_all_ch(funs(diff = .[segment==1]- .[segment==2]))
data %>% group_by(.sample_id) %>%  summarize_all_ch(funs(tvalue = t.test(.[segment==1], .[segment==2])$statistic))
data %>% group_by(.sample_id) %>%  summarize_all_ch(funs(diff = diff(.[segment==1], .[segment==2])))
```