bnicenboim/eeguana #3: Binding eeg_lsts with different channel names

auskate, 2018-10-17:
Just registering the old issue of binding with differently named EOG channels, which you say is due to metadata.

# recording with EOGV/EOGH
s3 <- read_vhdr("./3_preproc.vhdr", recording = 3)

# recording with VEOG/HEOG
s21 <- read_vhdr("./21-k_preproc.vhdr", recording = 21)

# bind
test <- rename(s21, EOGH  = HEOG, EOGV = VEOG) %>% 
  bind(., s3)

The last step gives some warning messages which later turn into problems with columns of different lengths. I had maybe a related issue later on when extracting and filtering a tibble, which I solved with droplevels() - although that's possibly a tidyverse quirk, not an eeguana one.

bnicenboim, 2018-10-18:
SOLVED!

so now you would do 
```r
channels_metadata <- channels_tbl(s3)
channels_tbl(s21) <- channels_metadata
```
And now they have the same channels metadat and you can bind them. If it worked please close it.

auskate, 2018-10-18:
Hmm. I read in the two different (now full) datasets:

data1 <- ...
data2 <- ...

then I did:

channels_metadata <- channels_tbl(data1)
channels_tbl(data2) <- channels_metadata

data <- bind(data1, data2)

Is that correct? b/c then I get the warning:

# Object size in memory 431 MB
Warning message:
Unknown channel in table of events 

When I look at data$events$.channels, the EOGV channel (only) appears randomly amongst the NAs.

Then later I get the same 'different lengths' errors when I try to join log files and/or segment.

bnicenboim, 2018-10-18:
do I have the problematic files? (or remind me the name of the file where this happens).

auskate, 2018-10-18:
yeah s21 and s3 are two of the problematic ones.

bnicenboim, 2018-10-18:
ok, now it should be working fine. But you got *warnings*, not errors. The objects were more or less ok, just some  events were off (which you probably don't need).

different lengths is another issue, and it's also a warning, it's because the join is creating discontinuous .ids. (I'll try to fix this issue as well)

bnicenboim, 2018-10-18:
ok, every issue was for completely different reasons. It seems that everything is working now. Please check it.

auskate, 2018-10-19:
Yes! Now it works 😃

closed 2018-10-19 by auskate