bnicenboim/eeguana #8: error when merging tibble with eeg_lst

auskate, 2018-11-12:
I get the following error when merging my log files with the eeg_lst:

Joining, by = c("recording", "segment")
Error in as_tibble.signal_tbl(data, .name_repair = "check_unique") : 
  unused argument (.name_repair = "check_unique")

Small script attached. Should work with data already in Dropbox/eegble_tests.

[merge_error.txt](https://github.com/bnicenboim/eeguana/files/2571607/merge_error.txt)

bnicenboim, 2018-11-12:
can you try it now?

auskate, 2018-11-12:
and I still get the same error :(

bnicenboim, 2018-11-12:
grrrr, now you should install a new version

auskate, 2018-11-12:
ok, new version installed. Error is new, but sounds simple...:

Joining, by = c("recording", "segment")
Error: 'semi_join_dt' is not an exported object from 'namespace:dplyr'

bnicenboim, 2018-11-13:
sorry, I was too cocky. Now it's tested, and it works in my computer, at least. Please check it.

auskate, 2018-11-13:
yep, works for me too!

closed 2018-11-13 by auskate