bnicenboim/eeguana #9: data.table warning when using plot_topo

auskate, 2018-11-12:
I get this warning when using plot_topo. Small script attached. It still works, so just fyi:

In addition: Warning message:
In `[.data.table`(extended_signal, , `:=`(.sample_id, sample_int(NA_integer_,  :
  Invalid .internal.selfref detected and fixed by taking a (shallow) copy of the data.table so that := can add this new column by reference. At an earlier point, this data.table has been copied by R (or was created manually using structure() or similar). Avoid key<-, names<- and attr<- which in R currently (and oddly) may copy the whole data.table. Use set* syntax instead to avoid copying: ?set, ?setnames and ?setattr. If this message doesn't help, please report your use case to the data.table issue tracker so the root cause can be fixed or this message improved.

[plot_topo_warning.txt](https://github.com/bnicenboim/eeguana/files/2571996/plot_topo_warning.txt)

bnicenboim, 2018-11-12:
yeah, I know. I get it too. I need to investigate further. But let's just ignore it for now :)

bnicenboim, 2020-06-26:
I don't get this anymore.

closed 2020-06-26 by bnicenboim