bnicenboim/eeguana #13: plot_topo issue

auskate, 2018-11-13:
Further to the email I sent you, I'm looking at the interpolation table produced by: 

eeg_sg %>% 
  filter(cond=="b"|cond=="d") %>%
  filter(between(as_time(.sample_id, unit = "milliseconds"), 80, 90)) %>%
  group_by(cond) %>%
  summarize_all_ch(mean, na.rm = TRUE) %>%
  interpolate_tbl()

but the channel labels are all NAs (shouldn't matter to the plot, but I can't tell from the xy coords which channel is which to check whether the amplitudes are right). Also, I get this warning when I run the above, which *may* have to do with the channel labels - but then there are also 12 NAs in each of the x & y coord variables:

Warning message:
In `[<-.factor`(`*tmp*`, !is_complete(data), value = "NA") :
  invalid factor level, NA generated

bnicenboim, 2018-11-13:
> Further to the email I sent you, I'm looking at the interpolation table
> produced by:
>
> eeg_sg %>%
> filter(cond=="b"|cond=="d") %>%
> filter(between(as_time(.sample_id, unit = "milliseconds"), 80, 90)) [...]
yeah, we are interpolating between the channels, so the interpolations
don't have a channel name. But the channel labels are not NA for the
original data points. So you can filter a channel, see the coordinates and
then check for the amplitudes near those coordinates

> Also, I get this warning when I run the above, which *may* have to do
> with the channel labels - but then there are also 12 NAs in each of the x &
> y coord variables:
The NA should correspond to the mastoids and eye electrodes, there
shouldn't be more than that. (These are not used in the interpolation, but
I left them there in case one wants to plot them).

> Warning message:
> In [<-.factor(*tmp*, !is_complete(data), value = "NA") :
> invalid factor level, NA generated

auskate, 2018-12-03:
Found my mistake - it was so dumb I'm not even going to post what it was XD

closed 2018-12-03 by auskate

bnicenboim, 2018-12-03:
Haha. Do the *plot_topo*s look good then? Have you compared them to the ones of fieldtrip or eeglab?

auskate, 2018-12-03:
They look great. I did one for the perm test in fieldtrip, but otherwise no. I will though! Visually, this one looks nicer though :)