bnicenboim/eeguana #35: mutate doesn't know how to use recently created arguments

auskate, 2018-12-04:
For example:

```{r}
# create time bins
eeg_topo <- eeg_tst %>%
  filter(cond %in% c("b","d")) %>%
  group_by(cond) %>%
  # it won't let me do these two mutates in one step:
  mutate(time = as_time(.sample_id, unit = "milliseconds")) %>%
  # 24 x 50ms bins in 1200ms:
  mutate(bin = ntile(time, 24))
```

(eeg_tst.rds is in Dropbox)

bnicenboim, 2018-12-04:
In the meanwhile, you can do this:
```
# create time bins
eeg_topo <- eeg_tst %>%
  filter(cond %in% c("b","d")) %>%
  group_by(cond) %>%
  # it won't let me do these two mutates in one step:
  mutate(bin = as_time(.sample_id, unit = "milliseconds") %>%
  # 24 x 50ms bins in 1200ms:
  ntile(time, 24))
```

bnicenboim, 2021-03-01:
See https://bnicenboim.github.io/eeguana/articles/gotchas.html

closed 2021-03-01 by bnicenboim