bnicenboim/eeguana #44: time in tibble and eeg_lst are inconsistent

bnicenboim, 2019-01-19:
```

data_1 <- eeg_lst(
  signal = signal_tbl(
    signal_matrix = as.matrix(
      data.frame(X = sin(1:30), Y = cos(1:30))
    ),
    ids = rep(c(1L, 2L, 3L), each = 10),
    sample_ids = sample_int(rep(seq(-4L, 5L), times = 3), sampling_rate = 500),
    dplyr::tibble(
      channel = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
      radius = NA, .x = c(1, 1), .y = NA_real_, .z = NA_real_
    )
  ),
  events = dplyr::tribble(
    ~.id, ~type, ~description, ~.sample_0, ~.size, ~.channel,
    1L, "New Segment", NA_character_, -4L, 1L, NA,
    1L, "Bad", NA_character_, -2L, 3L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 2L, "X",
    2L, "New Segment", NA_character_, -4L, 1L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 1L, "Y",
    3L, "New Segment", NA_character_, -4L, 1L, NA,
    3L, "Time 0", NA_character_, 1L, 1L, NA,
    3L, "Bad", NA_character_, 2L, 1L, "Y"
  ),
  segments = dplyr::tibble(.id = c(1L, 2L, 3L), recording = "recording1", segment = c(1L, 2L, 3L), condition = c("a","b","a"))
)
#just some different X and Y
data_2 <- mutate(data_1, recording = "recording2", X = sin(X +10), Y = cos(Y - 10), condition = c("b","a","b"))

data <- bind(data_1,data_2)

eeg_time <- mutate(data, time = as_time(.sample_id, unit = "seconds")) %>%
    summarize(mean = mean(time))
tbl_time <- data %>% as_tibble() %>%
    summarize(mean = mean(time))

test_that("as_time works as expected", {
  expect_equal(as.double(eeg_time$signal[["mean"]]), tbl_time$mean)
}
```

auskate, 2019-01-24:
Also, is it intended that as_time() only works within mutate()? E.g. I don't seem to be able to use it on its own:

``
data %>% as_time(.sample_id, "milliseconds")
``

bnicenboim, 2019-01-24:
The first argument has to be a  `sample_int` object, not necessarily inside mutate. E.g., `as_time(data$signal$.sample_id)`.

auskate, 2019-01-24:
Ah... I had also tried 

`as_time(data$signal[[".sample_id"]])`

and it didn't work, but I must have made a typo. Ignore!

bnicenboim, 2019-01-24:
> Ah... I had also tried
> 
> `as_time(data$signal[[".sample_id"]])`
> 
> and it didn't work, but I must have made a typo. Ignore!

this should work! I'll try it.

auskate, 2019-01-24:
no no it does work, i must have just made a typo on a previous attempt.

bnicenboim, 2019-01-27:
fixed in c2583867b3a1069a290224cc616a4be6a7d2c419 for experimental branch

commit c2583867 (2019-01-27): fixes #44
closed 2019-01-27 by bnicenboim