bnicenboim/eeguana #68: theme_eeguana(_empty) as exportable object

auskate, 2019-03-10:
Might be useful to have the default plot theme as a standalone plot function? (at least for testing :) ), e.g.:

```{r}
# create fake dataset
data_1 <- eeg_lst(
  signal = signal_tbl(
    signal_matrix = as.matrix(
      data.frame(X = sin(1:30), Y = sin(1:30))
    ),
    ids = rep(c(1L, 2L, 3L), each = 10),
    sample_ids = sample_int(rep(seq(-4L, 5L), times = 3), sampling_rate = 500),
    dplyr::tibble(
      channel = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
      radius = NA, .x = c(1, 1), .y = NA_real_, .z = NA_real_
    )
  ),
  events = dplyr::tribble(
    ~.id, ~type, ~description, ~.sample_0, ~.size, ~.channel,
    1L, "New Segment", NA_character_, -4L, 1L, NA,
    1L, "Bad", NA_character_, -2L, 3L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 2L, "X",
    2L, "New Segment", NA_character_, -4L, 1L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 1L, "Y",
    3L, "New Segment", NA_character_, -4L, 1L, NA,
    3L, "Time 0", NA_character_, 1L, 1L, NA,
    3L, "Bad", NA_character_, 2L, 1L, "Y"
  ),
  segments = dplyr::tibble(.id = c(1L, 2L, 3L),
                           recording = "recording1",
                           segment = c(1L, 2L, 3L),
                           condition = c("a", "b", "a"))
)

# just some different X and Y
data_2 <- mutate(data_1, recording = "recording2",
                 X = sin(X),
                 Y = sin(Y - .5),
                 condition = c("b", "a", "b"))

# bind it all together
data <- bind(data_1, data_2)

data %>%
  as_tibble() %>%
  ggplot2::ggplot(ggplot2::aes(time, amplitude)) + 
  ggplot2::geom_line(ggplot2::aes(group = .id, colour = condition), alpha = .5) +
  ggplot2::stat_summary(fun.y = "mean", geom = "line", aes(ggplot2::colour = condition), alpha = 1, size = 1) +
  ggplot2::facet_wrap(~ channel) +
  theme_eeguana()
# or
# eeguana::theme_eeguana()
```

bnicenboim, 2019-03-10:
you can do eeguana:::theme_eeguana

three : and no ()

auskate, 2019-03-11:
Fancy! Would it be worth it to have it the ggplot way as well though? E.g.
someone makes both default and tibble plots, and they want to make the
tibble plots match the default plots. Not sure whether it's really
necessary, was just a thought.