bnicenboim/eeguana #73: plot_topo give warnings

auskate, 2019-03-19:
It doesn't seem important (i.e. the plot works), but this warning comes up in testing: 
b.box[3] set to min y value

```{r}
library(eeguana)
library(testthat)

# create fake dataset
data_1 <- eeg_lst(
  signal = signal_tbl(
    signal_matrix = as.matrix(
      data.frame(X = sin(1:30), Y = sin(1:30))
    ),
    ids = rep(c(1L, 2L, 3L), each = 10),
    sample_ids = sample_int(rep(seq(-4L, 5L), times = 3), sampling_rate = 500),
    dplyr::tibble(
      channel = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
      radius = NA, .x = c(1, -10), .y = c(1, 1), .z = c(1, 10)
    )
  ),
  events = dplyr::tribble(
    ~.id, ~type, ~description, ~.sample_0, ~.size, ~.channel,
    1L, "New Segment", NA_character_, -4L, 1L, NA,
    1L, "Bad", NA_character_, -2L, 3L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 2L, "X",
    2L, "New Segment", NA_character_, -4L, 1L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 1L, "Y",
    3L, "New Segment", NA_character_, -4L, 1L, NA,
    3L, "Time 0", NA_character_, 1L, 1L, NA,
    3L, "Bad", NA_character_, 2L, 1L, "Y"
  ),
  segments = dplyr::tibble(.id = c(1L, 2L, 3L),
                           recording = "recording1",
                           segment = c(1L, 2L, 3L),
                           condition = c("a", "b", "a"))
)

# just some different X and Y
data_2 <- mutate(data_1, recording = "recording2",
                 X = sin(X),
                 Y = sin(Y - .5),
                 condition = c("b", "a", "b"))

# bind it all together
data <- bind(data_1, data_2)

# test topo plot
  expect_silent(data %>%
                  group_by(condition) %>%
                  summarise_all_ch(mean, na.rm = TRUE) %>%
                  plot_topo() + 
                  facet_grid(~ condition) +
                  annotate_head() +
                  geom_contour() +
                  geom_text(colour = "black")
                )
```

bnicenboim, 2019-04-11:
I don't get any warning. Could you try again?
But put the tribble of events inside an `events_tbl()` (events table have their own class now)

auskate, 2019-04-11:
I haven't seen this error recently tbh, it might be a previous-version
thing.

I'm running something large atm which has a few hours to go. Will double
check as soon as it's finished.

auskate, 2019-04-14:
I don't get the warning any more. I didn't get what you meant by the
`events_tbl()` bit though... do I just replace `dplyr::tribble` with
events_tbl? That gives me an error about the argument structure.

bnicenboim, 2019-04-14:
ah, no the you need to convert the tribble using as_events_tbl. I need to work on a more friendly way to convert objects. I'll just close this then.

closed 2019-04-14 by bnicenboim