bnicenboim/eeguana #74: plot_topo data is different length to eeg_interpolate_tbl

auskate, 2019-03-19:
Not sure if this matters, but putting it here anyway. The dataframes produced in plot_topo and eeg_interpolate_tbl are very different lengths:

Lengths differ: 400000 is not 8680

(is this related to the different electrode displays, i.e. polar vs. the other one?)

```{r}
library(eeguana)
library(testthat)
library(tidyverse)

# create fake dataset
data_1 <- eeg_lst(
  signal = signal_tbl(
    signal_matrix = as.matrix(
      data.frame(X = sin(1:30), Y = sin(1:30))
    ),
    ids = rep(c(1L, 2L, 3L), each = 10),
    sample_ids = sample_int(rep(seq(-4L, 5L), times = 3), sampling_rate = 500),
    dplyr::tibble(
      channel = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
      radius = NA, .x = c(1, -10), .y = c(1, 1), .z = c(1, 10)
    )
  ),
  events = dplyr::tribble(
    ~.id, ~type, ~description, ~.sample_0, ~.size, ~.channel,
    1L, "New Segment", NA_character_, -4L, 1L, NA,
    1L, "Bad", NA_character_, -2L, 3L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 2L, "X",
    2L, "New Segment", NA_character_, -4L, 1L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 1L, "Y",
    3L, "New Segment", NA_character_, -4L, 1L, NA,
    3L, "Time 0", NA_character_, 1L, 1L, NA,
    3L, "Bad", NA_character_, 2L, 1L, "Y"
  ),
  segments = dplyr::tibble(.id = c(1L, 2L, 3L),
                           recording = "recording1",
                           segment = c(1L, 2L, 3L),
                           condition = c("a", "b", "a"))
)

# just some different X and Y
data_2 <- mutate(data_1, recording = "recording2",
                 X = sin(X),
                 Y = sin(Y - .5),
                 condition = c("b", "a", "b"))

# bind it all together
data <- bind(data_1, data_2)

# test data doesn't change
  expect_equal(as.matrix(topoplot_eeg$data), 
               as.matrix(data %>%
                           group_by(condition) %>%
                           summarise_all_ch(mean, na.rm = TRUE) %>%
                           eeg_interpolate_tbl()))

```

bnicenboim, 2019-04-11:
`topoplot_eeg` is not there

bnicenboim, 2019-04-11:
I'm checking this in the test files, so don't do anything.
But THIS IS WEIRD!
I still haven't figure out why changing the coordinates does this mess....

bnicenboim, 2019-04-15:
ok, solved. It was because one of the electrodes was outside the area of interpolation in one projection and not in the other, plus the tables won't be the same because the internal one of the ggplot doesn't include the channel labels

closed 2019-04-15 by bnicenboim