bnicenboim/eeguana #106: documentation of filters

bnicenboim, 2019-07-11:
Link not working, better description of the edges vs cut off, transition width needs to be explicit (and maybe customizable) , phase needs to be explicit:

https://bnicenboim.github.io/eeguana/reference/filt.html

alexenge, 2020-06-26:
From a user's perspective, this one would be highly appreciated and make reporting a lot easier. Thanks for working on it!

bnicenboim, 2020-06-26:
Thanks for pointing this out, I think I just fixed this, in the next version this should be a bit better.

As you see there some issues still open for a long time. I don't have too much time to work on this package (at the moment), and I'm only trying to make it stable and to do what I need for my analysis (and a couple of my colleagues). 
If there is something urgent that you need, this is the moment to open issues or to comment on old ones :)
After the new version, I need to work on another project, and I won't fix anything that's not critical for some months.

alexenge, 2020-06-26:
Thanks, that was fast! The dplyr-related issue and the filter descriptions were the only two issues currently on my radar. I've recently been analysing a couple of data sets and I'm quite impressed by how stable and convenient the package is overall. Will definitely cite and recommend.

bnicenboim, 2020-06-26:
For now the transition width is customizable (and is reported) and I fixed the links. Eventually I'll do more there.

closed 2020-06-26 by bnicenboim