bnicenboim/eeguana #109: Error message: "could not find function select"

auskate, 2019-10-03:
Hey Bruno, I'm getting a weird error message when I try to use `select` to select a channel, even if I specify `eeguana::select`. I tried updating eeguana, still get the error. Here's an example with fake data:

```{r}
data_1 <- eeg_lst(
  signal_tbl =
    dplyr::tibble(X = sin(1:30), Y = cos(1:30),
                  .id = rep(c(1L, 2L, 3L), each = 10),
                  .sample = sample_int(rep(seq(-4L, 5L), times = 3), sampling_rate = 500)),
  channels_tbl = dplyr::tibble(
    .channel = c("X", "Y"), .reference = NA, theta = NA, phi = NA,
    radius = NA, .x = c(1, 1), .y = NA_real_, .z = NA_real_
  ),
  events_tbl = dplyr::tribble(
    ~.id, ~type, ~description, ~.initial, ~.final, ~.channel,
    1L, "New Segment", NA_character_, -4L, -4L, NA,
    1L, "Bad", NA_character_, -2L, 0L, NA,
    1L, "Time 0", NA_character_, 1L, 1L, NA,
    1L, "Bad", NA_character_, 2L, 3L, "X",
    2L, "New Segment", NA_character_, -4L, -4L, NA,
    2L, "Time 0", NA_character_, 1L, 1L, NA,
    2L, "Bad", NA_character_, 2L, 2L, "Y",
    3L, "New Segment", NA_character_, -4L, -4L, NA,
    3L, "Time 0", NA_character_, 1L, 1L, NA,
    3L, "Bad", NA_character_, 2L, 2L, "Y"
  ),
  segments_tbl = dplyr::tibble(.id = c(1L, 2L, 3L),
                               recording = "recording1",
                               segment = c(1L, 2L, 3L),
                               condition = c("a", "b", "a"))
)

data_1 %>% select(X)
```
I also got the attached error about backports.rdb when I created the fake data. I don't think it's relevant but I'm attaching it just in case.
[image]

bnicenboim, 2019-10-03:
What's the weird message?
Can you try with dplyr::select? tell me what happens, please.
Can you make another issue with backports error, could also post a reprex.
Thanks!

auskate, 2019-10-04:
The weird error message was "No tidyselect variables were registered", but the fake data example above is working today. Also, my own dataset where it wasn't working was maybe an older-style eeg_lst... I tried with a different dataset and everything is fine. 

The backports issue hasn't reoccurred, despite running the same code. Might have been a clash with some package I had loaded yesterday.... so I'll just close this. Sorry!

closed 2019-10-04 by auskate