bnicenboim/eeguana #121: import BDF

livioivil, 2020-03-25:
Hi notice an issue while importing a bdf file.
My example:
http://headit.ucsd.edu/recording_sessions/33a6adea-35ff-11e3-a2a9-0050563f2612
I downloaded: eeg_recording_1.bdf

When I tried to import with eeguana all was fine, but **the events table is empty**

Deg=eeguana::read_edf("eeg_recording_1.bdf")
pryr::object_size(Deg)
str(Deg)

While when I try with eegUtils it imported it well:

D0=eegUtils::import_raw("eeg_recording_1.bdf")
table(D0$ events$event_type)
length(D0$ events$event_type)
pryr::object_size(D0)

bnicenboim, 2020-03-25:
do you have some tool to cut the bdf and store it as bdf again with the same format?
It will be nice to have a very very short file with just some events so that I add it to the unit tests.

bnicenboim, 2020-03-25:
ok, nevermind, it's just the way BDF codes the triggers. I can test it with any BDF file. I'll solve it tomorrow.

livioivil, 2020-03-25:
Sorry,
I was not able to make a test file.
Don't stress too much about it.
I'll cope with it anyway

bnicenboim, 2020-03-26:
ok, it's working in the experimental version of eeguana:
```
remotes::install_github("bnicenboim/eeguana@experimental")
```
I'll do more tests and I'll release a new version

closed 2020-03-26 by livioivil