bnicenboim/eeguana #126: plot_ica error message

stonekate, 2020-04-11:
Hi Bruno,

I get an error (screenshot attached) when I try to plot ICA components. Below is the script I used, up until the point of the error. Data is in Dropbox.

```
s1 <- read_vhdr(file = "1.vhdr")

# filter
dataf <- eeg_filt_band_pass(s1, freq = c(.01, 30))

# only do ica on sentences (not breaks or questions)
eegsent <- eeg_segment(dataf, .description == "s195", end = .description == "s196") %>%
  # remove large artefacts first
  eeg_artif_minmax(-EOGH, -EOGV, threshold = 200, window = 200, unit = "ms")

# run ica
eegica <- eeg_ica(eegsent, -EOGH, -EOGV, -M1, -M2,
                  method = adapt_fast_ICA, ignore = .type == "artifact")

# plot_components(eegica)
eegica <- eegica %>%
  eeg_artif_step(EOGH, EOGV, threshold = 30, window = 200, unit = "ms", freq=c(1,10)) %>%
  eeg_artif_peak(EOGV, threshold=100, freq=c(1,10))

## view samples with blinks
s_peaks <- filter(events_tbl(eegica),  str_starts(.description, "peak")) %>% pull(.initial)

# this step throws an error
eeguana:::plot_ica.eeg_ica_lst(eegica, samples = seq(s_peaks[1]-2000,s_peaks[1]+2000))
```
[image]

bnicenboim, 2020-04-12:
ok, there are two problems, first:
You don't have sampels with the values `seq(s_peaks[1]-2000`, check what your minimum to start from.
And second your data doesn't have x,y,z coordinates for the channels

bnicenboim, 2020-04-12:
and by the way, this function is horrible. I need to write something better, and make it interactive with shiny. But this isn't going to happen in the near future :(, unless someone wants to contribute

closed 2020-04-12 by bnicenboim