bnicenboim/eeguana #129: chs_means() stopped working with dplyr version 1.0.0

alexenge, 2020-06-25:
Dear Bruno,

Thank you for your efforts with eeguana: The package is a pleasure to work with and a big step forward in making EEG data analysis more reproducible and less dependent on proprietary software!

I've just wanted to let you know that some of my previously written code stopped working after recently upgrading dplyr to version 1.0.0 (see https://www.tidyverse.org/blog/2020/06/dplyr-1-0-0/). It would be great if there was a way to (re-)integrate the two packages.

---

Here is a working example of `chs_means()` with dplyr version 0.8.5:

```r
dat <- read_vhdr("/Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.vhdr")
#> Reading file /Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.vhdr...
#> # Data from /Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.eeg was read.
#> # Data from 2 segment(s) and 64 channels was loaded.
#> # Object size in memory 439.2 Mb
dat <- dat %>% mutate(ROI = chs_mean(C1, Cz, C2))
"ROI" %in% channel_names(dat)
#> [1] TRUE
```
Whereas with dplyr version 1.0.0, I get the following error message:

```r
dat <- read_vhdr("/Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.vhdr")
#> Reading file /Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.vhdr...
#> # Data from /Users/alexander/Research/Aha/Aha1/Results/EEG/raw/Vp0001.eeg was read.
#> # Data from 2 segment(s) and 64 channels was loaded.
#> # Object size in memory 439.2 Mb
dat <- dat %>% mutate(ROI = chs_mean(C1, Cz, C2))
#> Error in `[.data.table`(.eeg_lst$.signal, segments): When i is a data.table (or character vector), the columns to join by must be specified using 'on=' argument (see ?data.table), by keying x (i.e. sorted, and, marked as sorted, see ?setkey), or by sharing column names between x and i (i.e., a natural join). Keyed joins might have further speed benefits on very large data due to x being sorted in RAM.
"ROI" %in% channel_names(dat)
#> [1] FALSE
```

Although the following workaround still does the trick, it would of course be great to get `chs_mean()` back (and potentially other dplyr-based functions which have stopped working with the new version).

```r
dat$.signal <- dat %>% signal_tbl %>% mutate(ROI = channel_dbl(rowMeans(dat %>% signal_tbl() %>% select(C1, Cz, C2))))
"ROI" %in% channel_names(dat)
#> [1] TRUE
```

Operative system (output of `sessionInfo()`):

```
R version 4.0.0 (2020-04-24)
Platform: x86_64-apple-darwin17.0 (64-bit)
Running under: macOS Catalina 10.15.5

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/4.0/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] eeguana_0.1.3.9000 dplyr_1.0.0        reprex_0.3.0      

loaded via a namespace (and not attached):
 [1] rstudioapi_0.11   knitr_1.29        magrittr_1.5      munsell_0.5.0     tidyselect_1.1.0  colorspace_1.4-1  R6_2.4.1          rlang_0.4.6      
 [9] tools_4.0.0       grid_4.0.0        gtable_0.3.0      data.table_1.12.8 xfun_0.15         htmltools_0.5.0   ellipsis_0.3.1    yaml_2.2.1       
[17] digest_0.6.25     tibble_3.0.1      lifecycle_0.2.0   crayon_1.3.4      ggplot2_3.3.2     purrr_0.3.4       vctrs_0.3.1       fs_1.4.1         
[25] glue_1.4.1        evaluate_0.14     rmarkdown_2.3     compiler_4.0.0    pillar_1.4.4      scales_1.1.1      generics_0.0.2    pkgconfig_2.0.3 
```

---

Thanks a lot!

Alexander

bnicenboim, 2020-06-25:
Thanks! Actually a ton of things stopped working for me! 
I'm on it, I'll probably update the package next week.

alexenge, 2020-06-25:
That's great to hear, thanks so much!

bnicenboim, 2020-06-26:
Can you try the dev version?
```
devtools::install_github("bnicenboim/eeguana@experimental")
```

It should be working there, I'll update the main version once I get rid of the new dplyr warnings...

alexenge, 2020-06-26:
Yes, `chs_mean()` (and re-naming channels using `mutate()`, which had also stopped working for me with dplyr 1.0.0), now seem to work fine in the dev version (without any warnings). Great!

closed 2020-06-26 by bnicenboim