bnicenboim/eeguana #133: Cannot install from CRAN

dipterix, 2020-07-18:
Cannot install from CRAN

```r
> install.packages('eeguana')
Installing package into ‘/Users/dipterix/Library/R/4.0/library’
(as ‘lib’ is unspecified)
Warning in install.packages :
  package ‘eeguana’ is not available (for R version 4.0.2)
```

Operative system:

Output of `sessionInfo()`:

```
R version 4.0.2 (2020-06-22)
Platform: x86_64-apple-darwin17.0 (64-bit)
Running under: macOS Catalina 10.15.5

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/4.0/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.5         digest_0.6.25      later_1.1.0.1      mime_0.9           R6_2.4.1           jsonlite_1.7.0    
 [7] backports_1.1.8    xtable_1.8-4       magrittr_1.5       RcppParallel_5.0.2 stringi_1.4.6      rlang_0.4.7       
[13] promises_1.1.1     startup_0.14.1     tools_4.0.2        stringr_1.4.0      shiny_1.5.0        parallel_4.0.2    
[19] httpuv_1.5.4       fastmap_1.0.1      compiler_4.0.2     dipsaus_0.0.9      base64url_1.4      htmltools_0.5.0   
```

bnicenboim, 2020-07-18:
It's not in Cran yet. You need devtools and to follow the instructions in the readme file.

closed 2020-07-18 by bnicenboim

dipterix, 2020-07-18:
Any plans to be on CRAN?

bnicenboim, 2020-07-18:
Eventually yes, but it needs some work. It has some "notes" that wouldn't allow it to pass CRAN. And I would like to submit to ropensci first also.