bnicenboim/eeguana #135: More control over plot_components()

themeo, 2020-09-26:
I can't force `plot_components()` to produce useful topoplots. Just to visualize this:

```r
dat_bline_ica %>% plot_components(standardize=F) + 
    scale_fill_gradientn(colors=rev(RColorBrewer::brewer.pal(n = 9, name = "RdBu")), limits=c(-3, 3), oob = scales::squish) + 
    theme(text=element_text(size=5))
```

Produces this figure: 
[image]

And when I resize it to a fullscreen figure, heads get out of proportion. 

It would be useful if there was a possibility to change the font size of `geom_text()` (`theme(text=...)` has no control over it), to remove electrode labels altogether, to enforce good proportions of the heads (always circular), to change the interpolation radius (sending `radius=1.4` doesn't work, sending `radius=2` plots the second IC). 

Sorry for so many issues in one report!

---

Output of `sessionInfo()`:

```
> sessionInfo()
R version 4.0.2 (2020-06-22)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19041)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252    LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] stringr_1.4.0      data.table_1.13.0  ggplot2_3.3.2      dplyr_1.0.2        eeguana_0.1.4.9000

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.5         prettyunits_1.1.1  ps_1.3.4           assertthat_0.2.1   rprojroot_1.3-2    digest_0.6.25      utf8_1.1.4         R6_2.4.1          
 [9] cellranger_1.1.0   backports_1.1.10   pillar_1.4.6       rlang_0.4.7        curl_4.3           readxl_1.3.1       rstudioapi_0.11    car_3.0-9         
[17] callr_3.4.4        desc_1.2.0         labeling_0.3       devtools_2.3.2     foreign_0.8-80     bit_4.0.4          munsell_0.5.0      MBA_0.0-9         
[25] compiler_4.0.2     pkgconfig_2.0.3    pkgbuild_1.1.0     tidyselect_1.1.0   tibble_3.0.3       rio_0.5.16         fansi_0.4.1        crayon_1.3.4      
[33] withr_2.3.0        grid_4.0.2         gtable_0.3.0       lifecycle_0.2.0    magrittr_1.5       scales_1.1.1       zip_2.1.1          cli_2.0.2         
[41] stringi_1.5.3      carData_3.0-4      farver_2.0.3       fs_1.5.0           remotes_2.2.0      testthat_2.3.2     ini_0.3.1          ellipsis_0.3.1    
[49] generics_0.0.2     vctrs_0.3.4        openxlsx_4.2.2     RColorBrewer_1.1-2 tools_4.0.2        forcats_0.5.0      bit64_4.0.5        glue_1.4.2        
[57] purrr_0.3.4        hms_0.5.3          processx_3.4.4     abind_1.4-5        pkgload_1.1.0      colorspace_1.4-1   sessioninfo_1.1.1  isoband_0.2.2     
[65] memoise_1.1.0      haven_2.3.1        usethis_1.6.3     
```

bnicenboim, 2020-09-26:
yeah, I didn't have time to work a lot on the plotting of ICA. (I'd like to eventually use shiny for this... ).
Would you mind sharing the data (or a subset of it)?

themeo, 2020-09-28:
This is absolutely not a priority, I usually draw ICs directly from EEGLAB when generating them with AMICA. I just thought I'd report those issues in case you weren't aware of them. 

Here are the files I used to generate the figure above: 
https://web.tresorit.com/l/0YncD#agxGH_S1oGXgxONvmAVvxQ

bnicenboim, 2020-09-28:
thanks!
I'll look at it in the next weeks.

bnicenboim, 2020-10-05:
ok, if I get it, you want that plot_components would behave like plot_topo, right?
where you can do the following:
```
dat_bline_ica %>%
       plot_topo() +
       # add a head and nose shape
       annotate_head() +
       # add contour lines
       geom_contour() +
       # add electrode labels
       geom_text(color = "black") +
       facet_grid(~condition)
```
is this right?

themeo, 2020-10-06:
Exactly, this sounds like a simple solution that would make things much easier because plot_topo() is much more configurable.

bnicenboim, 2020-10-09:
I fixed this, but I also did some changes that break things: changed a lot of arguments names. I'll need to test more and verify that I'm being consistent before I release a new version.
BTW, really cool that you're using the method function in "eeg_ica" :+1: 

If you're using the package, and you want to collaborate, drop me a line by email. I'm really busy now, and I'm doing basic maintenance, until we get new EEG data (from a pre-registration), and I doubt that will happen before March next year.

closed 2020-10-09 by bnicenboim