bnicenboim/eeguana #137: eeg_rereference() stopped working in the experimental branch

themeo, 2020-10-20:
```r
library(eeguana)

dat = read_vhdr("v1_CYOCM_Segmentation.vhdr")
dat %>% eeg_rereference(ref="A2")
```

leads to: 
> Error in \`[.data.table\`(signal, , \`:=\`((sel_ch), { : 
>  .SDcols should be column numbers or names

This is an example of a dataset: https://web.tresorit.com/l/Jre0A#3hj9ls3K7lFUPmQnXVQBwQ
but the error replicates for other datasets. Could result from the latest changes to dplyr verbs? I'm using eeguana from the experimental branch. The error doesn't occur when using the master branch. 

Output of `sessionInfo()`:

```
R version 4.0.2 (2020-06-22)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19041)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252    LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] eeguana_0.1.5.9000

loaded via a namespace (and not attached):
 [1] rstudioapi_0.11   magrittr_1.5      tidyselect_1.1.0  munsell_0.5.0     bit_4.0.4         colorspace_1.4-1  R6_2.4.1          rlang_0.4.7      
 [9] stringr_1.4.0     dplyr_1.0.2       tools_4.0.2       grid_4.0.2        data.table_1.13.2 gtable_0.3.0      ellipsis_0.3.1    bit64_4.0.5      
[17] tibble_3.0.4      lifecycle_0.2.0   crayon_1.3.4      ini_0.3.1         purrr_0.3.4       ggplot2_3.3.2     vctrs_0.3.4       glue_1.4.2       
[25] stringi_1.5.3     compiler_4.0.2    pillar_1.4.6      generics_0.0.2    scales_1.1.1      pkgconfig_2.0.3  
```

bnicenboim, 2020-10-20:
I'm slowly trying to be more consistent with the functions, so things might break :/ sorry
I think this is because it's now .ref
check the names of the arguments, I'm adding a dot everywhere except for the plot functions where I want to be similar to ggplot.

themeo, 2020-10-21:
Oh, I see, not a big problem, I checked that .ref works! The only thing is then the documentation may need a small overhaul at some point. Thank you!

closed 2020-10-21 by themeo

bnicenboim, 2020-10-21:
isn't there if you do `?eeg_rereference`?
But if you spot minor issues in documentation feel free to change them in the R files inside https://github.com/bnicenboim/eeguana/tree/master/R
You can edit the text and and do a pull request in the github site without too much git knowledge.

themeo, 2020-10-22:
Interesting, for some reason the documentation for
?eeg_rereference contains the version from the main branch despite the fact
that I installed the experimental version. Perhaps an install_github()
issue?

bnicenboim, 2020-10-22:
my bad, the Rd files were not updated. There devtools:document takes the info in the R files and created documentation (under /man/). 
I've updated the docs

themeo, 2020-10-22:
Awesome, thanks!