bnicenboim/eeguana #152: IIR filter error message: Column `iir` doesn't exist.

stonekate, 2021-02-12:
---

Hey, I'm trying to use the IIR filter but get the error above, maybe I missed something:

```r
# load data
dat <- read_vhdr("./1.vhdr")

# add channel positions
 channels_tbl(dat) <- select(channels_tbl(dat), .channel) %>%
    left_join(layout_32_1020)
  
  # filter
  datf <- eeg_filt_band_pass(dat, .freq = c(.01, 30), .method = "iir")
```

[image]

bnicenboim, 2021-02-12:
oh, maybe I didn't update the documentation. It should be `datf <- eeg_filt_band_pass(dat, .freq = c(.01, 30), .config = list(method = "iir"))`

stonekate, 2021-02-12:
yup that was it!

closed 2021-02-12 by stonekate