bnicenboim/eeguana #154: ICA error message post-IIR filtering

stonekate, 2021-02-12:
---
Hi again, now after I applied the IIR filters, ICA is giving me an error message:

[image]

rlang::last_error doesn't pick it up unfortunately, but I'll Dropbox you the dataset in case you need to recreate it. It happens at the final step of the pipeline below. I tried using FIR filters instead and ICA worked fine.

```r
  # load data
  dat <- read_vhdr("./1.vhdr")

  # add channel positions
  channels_tbl(dat) <- select(channels_tbl(dat), .channel) %>%
    left_join(layout_32_1020)
  

  # filter
  datf <- eeg_filt_band_pass(dat, .freq = c(.01, 30), .config = list(method = "iir"))
  
  
  # re-reference to the average of the mastoids
  dat_rr <- datf %>%
    eeg_rereference(-EOGH, -EOGV, .ref = c("M1", "M2"))
  

  # ICA: context start (s131) to sentence end (s132)
  dat_sglg <- eeg_segment(dat_rr, .description == "s131", .end = .description == "s132") %>%
    eeg_artif_minmax(-EOGH, -EOGV, .threshold = 100, .window = 150, .unit = "ms")

  dat_ica <- dat_sglg %>%
    eeg_ica(-EOGH, -EOGV, -M1, -M2, .method = adapt_fast_ICA, .ignore = .type == "artifact")

```

bnicenboim, 2021-02-12:
Cool and weird bug 😁
Have you tried with the other ica algorithms? It seems a problem of  the external ica function...

stonekate, 2021-02-13:
Hmm yes it seems so: I just checked both of the others and they work (fast_ICA, fast_ICA2)

bnicenboim, 2021-02-13:
Have you plotted the filtered data with iir vs fir? Is there something strange? 
Have you checked to do iir and them adapt ica with other subject data?

stonekate, 2021-02-14:
Yeah it happens with all the subjects.

Ah... definitely something strange in the plots :)

[image]

[image]

bnicenboim, 2021-02-14:
waw!
ok, thanks! 
I'll look at this during the week.

bnicenboim, 2021-02-15:
**tl;dr** This is signal's package fault, it will take me a while to fix, you can use a python-based alternative from R in the meanwhile (detailed below).

The issue is that there are two types of outputs for IIR filter design, second-order sections (‘sos’) form and  numerator/denominator form (‘ba’). `signal` from R only allows `ba`, which seem to suffer from numerical instabilities. Matlab and python also allow for `sos`, which is at least what the package MNE uses (by default).

So here there is a workaround if you're in a hurry, you install [mne](https://mne.tools/stable/install/mne_python.html#install-python-and-mne-python) in python. (If you don't have python also install it first). Then you install `reticulate` in R.

Now you can combine python and r, and specifically MNE and eeguana
```{r}
library(dplyr)
library(eeguana)
library(reticulate)
# use_condaenv("anaconda3") #maybe, not sure, or whatever conda environment you have installed if import mne in the following line gives an error
mne <- reticulate::import("mne")

datf <- dat %>% mutate_all(~ mne$filter$filter_data(., sfreq =512, l_freq = .01, h_freq = 30, method='iir') %>% 
channel_dbl)
# add channel positions, for some reason mutate_all looses that info
  channels_tbl(dat) <- select(channels_tbl(dat), .channel) %>%
    left_join(layout_32_1020)
  
```
And now you have filtered with MNE the eeguana object.

If you change the default output in MNE to the only output available in the signal package,  you actually get an error, instead of a crazy output:

```{r}
dat %>% mutate_all(~ mne$filter$filter_data(., sfreq =512, l_freq = .01, h_freq = 30, method='iir', iir_params = list(output = 'ba')) %>% channel_dbl)
## Error in py_call_impl(callable, dots$args, dots$keywords) : 
##  RuntimeError: Filter poles outside unit circle, filter will be unstable. Consider using different filter coefficients.
```

It will take me a while to fix this, if it can be even fixed. I see that there is an alternative to the package signal here
https://github.com/gjmvanboxtel/gsignal that has a `sosfilt` function. But I need to contact @gjmvanboxtel  (who happens to also work at Tilburg!), to see how mature gsignal is, and if it's a complete alternative to the signal package.

stonekate, 2021-02-15:
ooh, sorry it turned out complicated. I'm not in a hurry, but I'll try the MNE thing anyway.

bnicenboim, 2021-02-15:
no, don't apologize, it's great that you're discovering all the edge cases!

bnicenboim, 2022-06-13:
this should be now solved with new iir filters

closed 2022-06-13 by bnicenboim