bnicenboim/eeguana #164: drop_incomplete_segments() - new option to remove only if all channels are NA

themeo, 2021-04-07:
At present `drop_incomplete_segments()` drops the segment if any channel is NA. It would be useful to have a parameter to remove segments only if all channels are NA. 

so the logic would be the following:
```r
x %>% dplyr::group_by(.id) %>%
  dplyr::filter_at(channel_names(.),
                   dplyr::any_vars(!is.na(.)))
```

bnicenboim, 2021-04-07:
Sure, that seems straightforward.
But to be sure, the code that you posted works, right?

themeo, 2021-04-07:
Yes it does. :) It is based on your code, just changed all-not-NA to any-not-NA and got rid of `all()` function in the middle because in this case it was not necessary.

bnicenboim, 2021-04-08:
ok, then feel free to make a pull request to the experimental branch ;)
I guess there could be a parameter `all` set to `TRUE` by default.

Do you want to give it a try?

also I'm slowly making my internal functions to be based on data table, to speed up things, if you want to code it in data table, you're more than welcome.

themeo, 2021-04-10:
Sure, I'll be happy to give it a try after I'm done with moving.

bnicenboim, 2026-09-12:
if not please re-open

closed 2026-09-12 by bnicenboim