bnicenboim/eeguana #165: Zeros and extreme values in ICA

stonekate, 2021-06-15:
I'm getting some weird ICA results on datasets where it previously worked. I'm currently working in the experimental version eeguana_0.1.5.9000

There seem to be a lot of zeros and extreme values:

[image]

I tried both fast_ICA and adapt_fast_ICA since the first is the default and the second is used as an example in the documentation, but both results are the same. For one subject, I had to increase the number of iterations over 1000 for the ICA to converge (maxit argument of fast_ICA) - I haven't had to do that before, but it doesn't seem to be related to the above problem as it didn't help. 

```r
# segment and mark extreme amplitude changes
  dat_sglg <- eeg_segment(dat_rr, .description == "s131", .end = .description == "s132") %>%
    eeg_artif_minmax(-EOGH, -EOGV, .threshold = 100, .window = 150, .unit = "ms")

# ICA, ignoring marked artifacts
  dat_ica <- dat_sglg %>%
    eeg_ica(-EOGH, -EOGV, -M1, -M2, .method = adapt_fast_ICA, .ignore = .type == "artifact",
            .config = list(maxit = 1500))
```

bnicenboim, 2021-06-15:
ok, thanks for reporting that. I did quite a few untested changes in the experimental version. I'll check this ASAP. Please send me (a subset of) your dataset where this is happening

stonekate, 2021-06-15:
I just put in Dropbox: the same subject as in the picture here, filtered and ready for ica'ing with the code above (7_filtered.rds). In case you want a shorter dataset, there's a version of that file with just the first recording block (7_filtered_block1.rds). Then I added the output of the ICA that gave the picture above (7_ica.rds). Let me know if you want something else.

bnicenboim, 2021-06-16:
Could you try what happens with the master version of eeguana?
You can even install different versions side by side:
https://hansjoerg.me/2018/03/06/multiple-package-versions/

I want to know if I need to revert something, or if it was already broken

stonekate, 2021-06-16:
Oh no, I just tried, and the current master version gives the same result :(

It definitely did previously work!

stonekate, 2021-06-16:
Not sure if it offers any clues, but here is the ICA on the unfiltered data... still extreme, but at least not zero?

[image]

bnicenboim, 2021-06-16:
have you checked that the raw data before ICA looks ok?

stonekate, 2021-06-16:
yeah, the filtered data looks ok, there are a lot of big jumps in amplitude, but these should be marked and ignored according to the code above, right? This is the filtered data for the same subject:

[image]

bnicenboim, 2021-06-16:
they are not being ignored, that's the problem (not the ica algorithm by itself)
have you checked that this captures the artifacts?
```r
dat_sglg <- eeg_segment(dat_rr, .description == "s131", .end = .description == "s132") %>%
    eeg_artif_minmax(-EOGH, -EOGV, .threshold = 100, .window = 150, .unit = "ms")
```
and have you checked that .type is actually "artifact"?

stonekate, 2021-06-16:
hmm it seems to:

[image]

and here's the events table:

[image]

bnicenboim, 2021-06-16:
but those big jumps are captured right? they should be all in light pink

bnicenboim, 2021-06-16:
yes, play a bit with it, and then run eeg_events_to_NA before plotting to see if the large jumps disappear, ICA won't work with these large artifacts.

stonekate, 2021-06-16:
Ok. Did something change with the artifact marking? I swear I used it before on this exact dataset and I didn't notice anything weird!

bnicenboim, 2021-06-16:
I don't think so! But you said that the master version was behaving the same way, right?

stonekate, 2021-06-17:
Yeah but I had to update the master yesterday to get away from the experimental version (I now have side-by-side versions as per the link you sent - thanks). And it definitely worked in the master version I had before I updated.... unfortunately I don't know what previous version was or when I last updated it!

bnicenboim, 2021-06-17:
I don't do too many updates to the master version, maybe 2 or 3 last year....
In any case, I thought that the default should work fine, it's strange that it doesn't.
 I'll check it next week or the following, and I'll compare it with some other software (MNE, fieldtrip, etc). Is it ok, can it wait?

stonekate, 2021-06-17:
Yeah sure it's not urgent, in the meantime I'm trying different criteria for marking the artifacts in case it helps. I'll let you know if it does.

stonekate, 2021-06-21:
Minor update: shortening the window parameter of eeg_artif_minmax helped, but still did not pick up the most extreme jumps. I also installed all previous versions of eeguana and ran ICA, and got the same strange result as above, which is... very confusing... because remember I sent you a picture of the ERPs from the first subjects? I couldn't have made that if this had happened! I'm stumped.

bnicenboim, 2021-06-21:
ok, I'll investigate the minmax artifact detection, there must be something wrong there. 
Also I think that only ERPlab uses this artifact detection method (I need to check it), why are you using this specific artifact detection before ICA? (Or was my idea :) ?)
I think that there are a couple of new papers with guidelines for EEG analysis, maybe we should check what they recommend before ICA.

stonekate, 2021-06-22:
I'm just following your idea dude :) It seems reasonable.

This paper seems potentially useful: https://www.sciencedirect.com/science/article/pii/S1053811919307086?via%3Dihub#sec2
He found that high pass filtering (no low pass) + "overweighting" eye artifacts before ICA was the best (there's github code here https://github.com/olafdimigen/opticat although I can't find how the overweighting was implemented... might be an EEGlab function). 

Steve Luck is apparently writing a whole chapter on ICA, but it's not out yet.

bnicenboim, 2021-06-22:
> Steve Luck is apparently writing a whole chapter on ICA, but it's not out yet.
yeah, I probably implemented what he suggested somewhere. I *think* it did what ERPLAB does, but I'm not sure :), I need to check...

Also Manylabs has standardized pipeline: https://psyarxiv.com/528nr/ 

> (there's github code here https://github.com/olafdimigen/opticat although I can't find how the overweighting was implemented... might be an EEGlab function).
I'll check it as well. Translating eeglab functions is relatively straightforward. But it's important to find out how well tested is this, and how standard it is.

Anyways, I have work to do here it seems :)

stonekate, 2021-06-22:
OK, maybe I found the problem: there are some very bad segments that are (understandably?) not picked up by the minmax or step functions, and possibly never would be no matter what you put in .window or .threshold:
[image]

(btw this image is from a shiny app I made for scrolling through segments - it's extremely basic and clunky and slow, but I'll put it in dropbox in case it's useful)

If I chop out the very bad segments (31:34, 210), the ICA now looks pretty good:
[image]

I worried though that I'm marking artifacts which include blinks and then telling ICA to ignore these... but doesn't ICA need that information? 
[image]

Anyway, I tried ICA *without* ignoring the artifacts and it looks kind of the same:
[image]

So in summary, is there some way to auto-mark the bad stuff in the first picture?

bnicenboim, 2021-06-23:
yes, marking too many artifacts is bad, the idea is to only remove the ones that are really big and break the ICA, like the ones we are missing right now with the min-max

bnicenboim, 2024-03-12:
I think there's nothing for me to do here

closed 2024-03-12 by bnicenboim