bnicenboim/eeguana #167: Feature request: eeg_events_to_NA for single channel

stonekate, 2021-07-06:
There is currently an option in eeg_events_to_NA to set .all_chs to FALSE, but I can't figure out how to make it NA just one channel. E.g. when eyeballing the EEG, I see one electrode is drifting but I don't want to throw out the whole segment. So I want to manually set its .value to NA, something like:

```
dat_seg %>%
  eeg_events_to_NA(.channel == "P3" & .id %in% c(61:62), .all_chs = FALSE, .entire_seg = FALSE, .drop_events = FALSE)
```

Maybe it's already possible? If not, could it be? 

Actually, I looked through the code for the function and tested its effects on the signal_tbl, and I'm not sure what .all_chs = FALSE is doing... there's if(.all_chs), then do blah blah blah, but no "else".

bnicenboim, 2021-07-06:
if `.all_chs = FALSE` it should do that, it doesn't? so is`.all_chs` not doing anything?

stonekate, 2021-07-06:
I think so :)

stonekate, 2021-07-06:
but it doesn't accept a .channel == argument in the first position no matter what I do, because I think there is no channel information in the events table (it's all NAs for me in the channel column anyway)

bnicenboim, 2021-07-06:
ok, maybe send me a reprex? with a reduced dataset

stonekate, 2021-07-06:
in Dropbox (or will be shortly), see reprex_eeg_eventstoNA.R

bnicenboim, 2022-02-03:
With a lot of delay, a solution is to add the channel name that you want to remove to the events table.
In any case, I'm working again with eeguana, and I know what I mean. Very late, but I'll fix it....

bnicenboim, 2022-03-11:
ok, I'll close it, eeg_events_to_NA is supposed to look at the events table. This shouldn't work:
```
dat_seg %>%
  eeg_events_to_NA(.channel == "P3" & .id %in% c(61:62), .all_chs = FALSE, .entire_seg = FALSE, .drop_events = FALSE)
```
This is the way you are supposed to do it.

```
eeg_fake <- eeg_lst(signal_tbl = tibble(.id = rep(1:10, each =100),
                                        .sample = sample_int(rep(1:100, 10),500),  
                                        P1 = channel_dbl(rnorm(1000)), 
                                        P3 = channel_dbl(rnorm(1000))))

eeg_mutate(eeg_fake, P3 = case_when(.id %in% 1:5 ~ NA, 
                                    TRUE ~ P3))
```

bnicenboim, 2022-03-11:
I added this here https://github.com/bnicenboim/eeguana/discussions/182, for future reference.

closed 2022-03-11 by bnicenboim