bnicenboim/eeguana #193: Add ignore argument to eeg_segment

stonekate, 2022-04-27:
For example, if I want to pull out the word after the target segment I can do:

```
eeg %>%  eeg_segment(lead(.description == "S100"), lim = c(-200, 1000), .unit = "milliseconds") 
```

but BV-preprocessed data marks UserDefined events in the events table, like Blink, so sometimes the event after S100 might be a Blink. Some way to ignore non-stimulus events would be nice? Or maybe there's already a way to do this...

bnicenboim, 2022-04-27:
Why can't you just do the following?
```
eeg %>%  eeg_segment(lead(.description == "S100") & .description != "Blink", lim = c(-200, 1000), .unit = "milliseconds") 
```
The `...` works exactly as `dplyr::filter` but since all the conditions go together in one argument, you need to do and with `&` rather than with `,`

stonekate, 2022-05-02:
Ok yeah that works totally fine, I'll close this, thx!