bnicenboim/eeguana #194: error message IIR filter

stonekate, 2022-05-30:
In both stable and experimental eeguanas I get the following error with the pipeline below, just for one single subject, all other subjects are fine:

```
# Load data
dat <- read_vhdr("./2.vhdr")

# Add channel positions
channels_tbl(dat) <- select(channels_tbl(dat), .channel) %>%
    left_join(layout_32_1020)
  
# IIR filter
datf <- eeg_filt_band_pass(dat, .freq = c(.01, 30), .config = list(method = "iir"))

```

[image]

bnicenboim, 2022-06-08:
just FYI, I'm working on it. I'm changing all the filtering to a newer package, I think I'll manage to finish it next week

bnicenboim, 2022-06-13:
ok, I think I fixed it. Please check it with the new experimental version.

closed 2022-06-14 by bnicenboim