bnicenboim/eeguana #195: IIR filter not behaving as expected

stonekate, 2022-05-30:
So the IIR filter is doing something weird after all. With the following pipeline, I get extreme values in the signal table (both stable and experimental eeguanas):

```
# Load data
dat <- read_vhdr("./1.vhdr")

# Add channel positions
channels_tbl(dat) <- select(channels_tbl(dat), .channel) %>%
    left_join(layout_32_1020)
  
# IIR filter
dat <- eeg_filt_band_pass(dat, .freq = c(.01, 30), .config = list(method = "iir"))

# Re-reference
dat <- dat %>%
    eeg_rereference(-HEOG, -VEOG, .ref = c("M1", "M2"))

# Segment context start (s131) to sentence end (s132)
dat_seg <- eeg_segment(dat, .description == "s131", .end = .description == "s132")

#  Plot
plot(filter(dat_seg, .id %in% c(10)))
```

[image]

bnicenboim, 2022-06-13:
hi, please check it now.

stonekate, 2022-06-14:
looks good now!

closed 2022-06-14 by bnicenboim