bnicenboim/eeguana #204: Error when binding individual eeg_lsts together

stonekate, 2022-07-12:
Hi Bruno, I'm having trouble binding specific participants together (I've put the problematic ones in dropbox). I've tried both main and experimental eeguanas. All the others bind fine so I don't think it's a bug, I just don't understand the error messages from data.table. Any ideas?

```
eeg15 <- readRDS("eeg.15.rds")
eeg16 <- readRDS("eeg.16.rds")
eeg17 <- readRDS("eeg.17.rds")
eeg18 <- readRDS("eeg.18.rds")
eeg19 <- readRDS("eeg.19.rds")

bind(eeg15, eeg16) # works fine
bind(eeg16, eeg17) # this and the following get the error message below
bind(eeg16, eeg18) 

```

I don't know what this means :(
[image]

```
# eeg19 has only RM as a reference because LM didn't work (the others are re-referenced to the average of LM/RM) so it gets an extra warning about that, but I think the error is the same as above:
bind(eeg16, eeg19) 
```
[image]

stonekate, 2022-07-12:
PS. All of is_channel_dbl() in the signal tables = TRUE

bnicenboim, 2022-07-13:
the problem is objects saved with different versions of eeguana. Use the latest experimental version and do `as_eeg_lst(eegN)` before you bind them

closed 2022-07-13 by bnicenboim

stonekate, 2022-07-13:
aah it worked, thank you!