bnicenboim/eeguana #210: case_when maybe not working with eeg_mutate

stonekate, 2022-09-26:
A weird one:

Previously this worked to NA an electrode for a particular participant:

```r
eeg <- eeg_mutate(eeg, AFz = case_when(.recording == 3 ~ NA))
```

But in recent eeguana versions (at least experimental and the last master), it NAs that electrode for every participant. This works as expected however:

```
eeg <- eeg_mutate(eeg, AFz = ifelse(.recording == 3, NA, AFz))
```

I can't imagine why specifically `case_when()` wouldn't work, but maybe it makes sense to you.

bnicenboim, 2022-10-10:
again, this doesn't seem to be eeguana:

``` r
library(dplyr)
#> 
#> Attaching package: 'dplyr'
#> The following objects are masked from 'package:stats':
#> 
#>     filter, lag
#> The following objects are masked from 'package:base':
#> 
#>     intersect, setdiff, setequal, union
df <- data.frame(a= 1:10, b= c(rep(1,5),rep(2,5)))
df
#>     a b
#> 1   1 1
#> 2   2 1
#> 3   3 1
#> 4   4 1
#> 5   5 1
#> 6   6 2
#> 7   7 2
#> 8   8 2
#> 9   9 2
#> 10 10 2
df %>% mutate(a = case_when(b == 1 ~ NA))
#>     a b
#> 1  NA 1
#> 2  NA 1
#> 3  NA 1
#> 4  NA 1
#> 5  NA 1
#> 6  NA 2
#> 7  NA 2
#> 8  NA 2
#> 9  NA 2
#> 10 NA 2
df %>% mutate(a = ifelse(b == 1, NA, a))
#>     a b
#> 1  NA 1
#> 2  NA 1
#> 3  NA 1
#> 4  NA 1
#> 5  NA 1
#> 6   6 2
#> 7   7 2
#> 8   8 2
#> 9   9 2
#> 10 10 2
```

<sup>Created on 2022-10-10 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

it's the same issue, right?

I'm starting to hate dplyr

stonekate, 2022-10-10:
oh that's really concerning, I use case_when everywhere :/

bnicenboim, 2022-10-10:
Only if you do this:
```
df %>% mutate.(a = case_when(b == 1 ~ NA_real_,
                             TRUE ~ a))
```
 it works. :-o

I'm just happy, it's not my fault :dancers: 

Closing

closed 2022-10-10 by bnicenboim