bnicenboim/eeguana #213: Error when marking artifacts

stonekate, 2022-10-19:
Hey Bruno, 

I have gotten a novel error when marking artifacts and am not sure how to interpret it (I don't think it's from eeguana, but I'm not sure what eeguana does that elicits it). Any ideas? (this is not the INHIBIT data and is separate to the other issue I'm working on). It's only specifically for this one subject and only since the last eeguana update.

```r
dat_ica <- readRDS("Dropbox/eeg_data_kate/5_ica.rds")

dat_seg <- dat_ica %>%
          eeg_segment(lead(.description %in% c("S200"), 2) & .type == "Stimulus", .lim = c(-0.2, 1))

dat_art <- dat_seg %>%
      eeg_artif_minmax(-HEOG, -VEOG, -LO1, .threshold = 100, .window = 150, .unit = "ms") %>%
      eeg_artif_step(-HEOG, -VEOG, -LO1, .threshold = 50, .window = 150, .unit = "ms")

```

[image]

bnicenboim, 2022-10-19:
it seems this is eeguana's issue. Is the file in the dropbox? I'll look at it

stonekate, 2022-10-19:
oh, yep it's there

stonekate, 2023-01-26:
In the end I think this was my issue - I think there was a missing end-of-sentence marker for one sentence so that .id had a .sample marked NA. When eeg_artif_minmax tried to find the min() and max() .sample, it failed because the vector contained an NA. I don't think you need to do anything.

closed 2023-01-26 by stonekate