bnicenboim/eeguana #221: Removing bad ICA components not correcting blinks in new data, same recording system

stonekate, 2023-07-31:
Hey Bruno,

I'm preprocessing some new data and strangely the ICA doesn't seem to be working... or maybe I'm insane... I have a few good subjects with really clear blink components and yet the blinks are not corrected/removed at all (as far as I can tell??). I put a script and some data in Dropbox/kate_eeg_data. I put the whole preprocessing pipeline but you can skip to line 280 to load and plot the should-be-already-blink-corrected example. The ICA components can be plotted from line 144.

I double-checked my code against your website example, but probably I'm just missing something...

Thanks,
Kate

bnicenboim, 2023-08-08:
I'll be away until the end of august and then I'll check it. Maybe some other package changed in the meanwhile.

stonekate, 2023-08-08:
No hurry. It's also a bit weird, and specific to one dataset. It's probably my issue but I can't find what I might have done to cause it.

stonekate, 2023-08-14:
Finally spotted my veeeery dumb mistake... nothing for you to check here!

closed 2023-08-14 by stonekate