bnicenboim/eeguana #222: 'mutate.' is not an exported object from 'namespace:tidytable'

spressi, 2023-09-18:
Tried to use eeguana on a new install of R 4.3.1 (2023-06-16 ucrt) with RStudio 2023.06.2. 
I ran into a small problem with the dependent package `gsignal` that could be fixed by manually installing it prior to the call to `devtools::install_github("bnicenboim/eeguana")`. When trying to open a file now with `read_vhdr`, however, it outputs the error message: `Error: 'mutate.' is not an exported object from 'namespace:tidytable'`. Seems like a problem with an updated version of tidyverse / dplyr?

```r
files = list.files(path="data/", pattern="_1.vhdr")
file = sample(files, 1) #test subject
file %>% eeguana::read_vhdr()
```

*Operative system:*

(Windows 10, Ubuntu 20.04.2 LTS, etc) 

*Output of `sessionInfo()`:*

```
R version 4.3.1 (2023-06-16 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19045)

Matrix products: default

locale:
[1] LC_COLLATE=German_Germany.utf8  LC_CTYPE=German_Germany.utf8    LC_MONETARY=German_Germany.utf8 LC_NUMERIC=C                   
[5] LC_TIME=German_Germany.utf8    

time zone: Europe/Berlin
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
 [1] lubridate_1.9.2 forcats_1.0.0   stringr_1.5.0   dplyr_1.1.3     purrr_1.0.2     readr_2.1.4     tidyr_1.3.0     tibble_3.2.1   
 [9] ggplot2_3.4.3   tidyverse_2.0.0

loaded via a namespace (and not attached):
 [1] utf8_1.2.3         generics_0.1.3     stringi_1.7.12     hms_1.1.3          digest_0.6.33      magrittr_2.0.3     grid_4.3.1        
 [8] timechange_0.2.0   pkgload_1.3.2.1    fastmap_1.1.1      rprojroot_2.0.3    processx_3.8.2     sessioninfo_1.2.2  pkgbuild_1.4.2    
[15] urlchecker_1.0.1   ps_1.7.5           promises_1.2.1     tidytable_0.10.1.9 fansi_1.0.4        scales_1.2.1       cli_3.6.1         
[22] shiny_1.7.5        crayon_1.5.2       rlang_1.1.1        ellipsis_0.3.2     munsell_0.5.0      remotes_2.4.2.1    withr_2.5.0       
[29] cachem_1.0.8       devtools_2.4.5     tools_4.3.1        tzdb_0.4.0         memoise_2.0.1      colorspace_2.1-0   httpuv_1.6.11     
[36] eeguana_0.1.8.9001 curl_5.0.2         vctrs_0.6.3        R6_2.5.1           mime_0.12          lifecycle_1.0.3    htmlwidgets_1.6.2 
[43] fs_1.6.3           usethis_2.2.2      miniUI_0.1.1.1     desc_1.4.2         pkgconfig_2.0.3    callr_3.7.3        pillar_1.9.0      
[50] later_1.3.1        gtable_0.3.4       profvis_0.3.8      data.table_1.14.8  glue_1.6.2         Rcpp_1.0.11        tidyselect_1.2.0  
[57] rstudioapi_0.15.0  xtable_1.8-4       htmltools_0.5.6    compiler_4.3.1     prettyunits_1.1.1 
```

bnicenboim, 2023-09-19:
Thanks! An update is super overdue. I'll take care of this by the end of October.

(But it could be that I already started to fixed the dev version, but I didn't finish, you can try it or wait a bit).

bnicenboim, 2024-01-26:
it took me a while, but now it shouldn't give any problem...

closed 2024-03-12 by bnicenboim