bnicenboim/eeguana #223: artifact detection adds new .id values?

spressi, 2023-09-22:
I need to perform an artifact detection on a **subset** of the trials of my EEG data (I want to save computation time because I plan to run a multiverse analysis). I cannot segment only for the relevant part of the data because I need the segment .id to align with the trial number (I have 432 trials per subject, only need to analyse 288 per subject, but the position of these 288 trials is randomized).

I used something like this to achieve my goal:
```r
eeg.artifacts = eeg.segmented %>% eeg_filter(.id %in% trials.analyse) %>% 
    eeg_artif_minmax(contains("EOG"), .threshold=60, .window=.5, .unit="s") %>% #too much activity in EOG
    eeg_artif_minmax(!contains("EOG") & !contains("IO"), .threshold=100, .window=.5, .unit="s") #too much activity in EEG
```
Unexpectedly, `eeg.artifacts` has entries in the .signal and `events_tbl` that should not be present according to the `eeg_filter`.
More specifically, `eeg.artifacts$.signal$.id %>% max()` turns out to be 432 even though its predecessor `eeg.segmented %>% eeg_filter(.id %in% trials.analyse) %>% events_tbl() %>% pull(.id) %>% max()` is 431 (same if I check the .id of the .signal table after `eeg_filter`).
Could it be that the `eeg_artif_minmax` (and probably other artifact detection functions) do not use the .id of the current segment but count upwards or expect that all ids are present or something like that?

---
Reprex

```r
#load and segment some EEG data called eeg.segmented
ids.all = eeg.segmented %>% events_tbl() %>% pull(.id) %>% unique()
trials.analyse = sample(ids.all, ids.all %>% length() %>% {. / 2} %>% round())
eeg.artifacts = eeg.segmented %>% eeg_filter(.id %in% trials.analyse) %>% 
    eeg_artif_minmax(contains("EOG"), .threshold=60, .window=.5, .unit="s") %>% #too much activity in EOG
    eeg_artif_minmax(!contains("EOG") & !contains("IO"), .threshold=100, .window=.5, .unit="s") #too much activity in EEG
setdiff(eeg.artifacts %>% events_tbl() %>% pull(.id) %>% unique(), trials.analyse)
```

*Operative system:*

(Windows 10, Ubuntu 20.04.2 LTS, etc) 

*Output of `sessionInfo()`:*

```
R version 4.2.1 (2022-06-23 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19045)

Matrix products: default

locale:
[1] LC_COLLATE=German_Germany.utf8  LC_CTYPE=German_Germany.utf8    LC_MONETARY=German_Germany.utf8 LC_NUMERIC=C                   
[5] LC_TIME=German_Germany.utf8    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
 [1] forcats_0.5.2      stringr_1.4.1      dplyr_1.0.9        purrr_0.3.4        readr_2.1.2        tidyr_1.2.0        tibble_3.1.8      
 [8] ggplot2_3.3.6      tidyverse_1.3.2    eeguana_0.1.8.9001

loaded via a namespace (and not attached):
 [1] tidyselect_1.1.2    haven_2.5.1         gargle_1.2.0        colorspace_2.0-3    vctrs_0.4.1         generics_0.1.3     
 [7] gsignal_0.3-5       utf8_1.2.2          rlang_1.0.4         pracma_2.3.8        pillar_1.8.1        glue_1.6.2         
[13] withr_2.5.0         DBI_1.1.3           tidytable_0.8.1     dbplyr_2.2.1        modelr_0.1.9        readxl_1.4.1       
[19] lifecycle_1.0.1     munsell_0.5.0       gtable_0.3.0        cellranger_1.1.0    rvest_1.0.3         labeling_0.4.2     
[25] tzdb_0.3.0          fansi_1.0.3         broom_1.0.1         Rcpp_1.0.9          backports_1.4.1     scales_1.2.1       
[31] googlesheets4_1.0.1 jsonlite_1.8.0      farver_2.1.1        fs_1.5.2            digest_0.6.29       hms_1.1.2          
[37] stringi_1.7.8       RcppRoll_0.3.0      grid_4.2.1          cli_3.3.0           tools_4.2.1         magrittr_2.0.3     
[43] crayon_1.5.1        pkgconfig_2.0.3     MASS_7.3-58.1       ellipsis_0.3.2      groundhog_3.1.1     data.table_1.14.2  
[49] xml2_1.3.3          reprex_2.0.2        googledrive_2.0.0   lubridate_1.8.0     assertthat_0.2.1    httr_1.4.4         
[55] rstudioapi_0.14     R6_2.5.1            compiler_4.2.1  
```

bnicenboim, 2023-09-22:
Could you share some data to reproduce the example?

spressi, 2023-09-22:
Sorry, now I cannot reproduce it myself anymore. Seems like I should have checked more thoroughly before posting, my bad. I'll try again within the next few days and will reopen the issue if I can reproduce the problem.

closed 2023-09-22 by spressi