bnicenboim/eeguana #236: could not find function "across."

prattems, 2024-03-08:
Several functions throw this error including eeg_rereference() and eeg_filt_band_pass(). I'm loading eeguana (version 0.1.10.9001) followed by dplyr (version 1.1.4). It looks like mutate.() is defined internally, but I'm not sure where across.() should be coming from. 

Thanks so much, for the help and this package! I've only started using it but it's going to be terribly helpful (if I can learn dplyr). 

Mike

bnicenboim, 2024-03-09:
Hi, thanks for reporting this.
Could you send me reproducible example? if you install, reprex (https://reprex.tidyverse.org/), you can use the Addins menu in Rstudio and do Reprex selection.
I want to know the order of loading packages and what exactly you're doing with the functions

prattems, 2024-03-11:
Thanks so much,

In the example below I use your builtin data and code from the help files. The call to eeg_segment works but the subsequent call to eeg_reference gives the error regarding across.() not being found. I find the same error if dplyr is loaded before eeguana.

``` r
library(eeguana)
#> eeguana version 0.1.10.9001
#> An introduction to the package can be found in https://bruno.nicenboim.me/eeguana/articles/intro.html
library(dplyr)
#> 
#> Attaching package: 'dplyr'
#> The following objects are masked from 'package:eeguana':
#> 
#>     across, between, c_across
#> The following objects are masked from 'package:stats':
#> 
#>     filter, lag
#> The following objects are masked from 'package:base':
#> 
#>     intersect, setdiff, setequal, union

data_faces_10_trials %>% eeg_segment(.description %in% c("s70", "s71"),
                                     .lim = c(-5, 1))
#> # Total of 9 segments found.
#> # Object size in memory 7.3 Mb after segmentation.
#> # EEG data:
#> 
#> # Signal table:
#> Key: <.id, .sample>
#>          .id      .sample           Fp1           Fpz           Fp2
#>        <int> <sample_int> <channel_dbl> <channel_dbl> <channel_dbl>
#>     1:     1        -2499    -21.109169    -28.948433   -25.2941818
#>     2:     1        -2498    -26.754496    -31.574997   -29.3033142
#>     3:     1        -2497    -28.524906    -30.061056   -29.8119774
#>     4:     1        -2496    -19.430933    -21.690756   -22.9545746
#>     5:     1        -2495     -5.725405    -12.675667   -14.1778440
#>    ---                                                             
#> 27005:     9          497     -8.819434    -10.321518    -4.1491461
#> 27006:     9          498     -6.959885    -10.197778    -3.9415770
#> 27007:     9          499     -1.756994     -7.125042    -1.8327932
#> 27008:     9          500      3.828817     -3.076797     0.1685476
#> 27009:     9          501      6.447233     -1.101932     0.8303175
#>                   F7            F3            Fz            F4            F8
#>        <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl>
#>     1:  -10.91393471    -7.5658569    -8.7408314     -8.170362    -6.2060385
#>     2:  -15.56297112    -8.5204601    -8.6705694     -7.897309    -6.2139606
#>     3:  -18.79760742   -10.9199018    -8.0307159     -6.302173    -7.0704432
#>     4:  -14.15851784   -11.1185970    -5.9641218     -3.515551    -9.4960175
#>     5:   -2.97854924    -7.5502534    -3.7647977     -1.728970   -11.8486443
#>    ---                                                                      
#> 27005:   -0.04505777    -3.4396243    -0.3492746      1.448371    -0.9764605
#> 27006:   -0.39633083    -5.2603526    -0.5973411      2.423673    -1.4587488
#> 27007:    1.76945877    -4.0067720     0.8627796      3.911339    -2.3904738
#> 27008:    5.14945936    -1.2876422     2.6780972      4.573048    -3.9883673
#> 27009:    7.85265970     0.7570639     3.2915244      3.903461    -5.8100514
#>                  FC5           FC1           FC2           FC6            M1
#>        <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl>
#>     1:   -10.3604956     -8.741921    -11.687920    -3.2395220   -1.33440495
#>     2:    -8.3586874     -9.393848    -11.007945    -2.0159216   -0.40914702
#>     3:    -6.6766038    -10.921223     -9.508024     0.6382501   -0.02219653
#>     4:    -5.3997726    -11.370429     -8.181183     2.3170936   -0.15380359
#>     5:    -1.5244772    -10.140000     -8.618706     0.1669424   -0.42611933
#>    ---                                                                      
#> 27005:    -3.1192555      7.583683      4.683777     2.0146487   -0.75850582
#> 27006:    -5.0121384      7.003987      5.093409     2.6218574   -0.48889470
#> 27007:    -3.8819036      8.010637      6.655658     2.4396117    0.37114978
#> 27008:     0.2056897      9.351544      7.915032     1.1819088    0.79920685
#> 27009:     3.5806384      9.694896      7.764079    -0.7410481    0.47170472
#>                   T7            C3            Cz            C4            T8
#>        <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl> <channel_dbl>
[... 110 lines cut ...]
#>  [5] LC_MONETARY=en_US.UTF-8    LC_MESSAGES=en_US.UTF-8   
#>  [7] LC_PAPER=en_US.UTF-8       LC_NAME=C                 
#>  [9] LC_ADDRESS=C               LC_TELEPHONE=C            
#> [11] LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C       
#> 
#> attached base packages:
#> [1] stats     graphics  grDevices utils     datasets  methods   base     
#> 
#> other attached packages:
#> [1] dplyr_1.1.4         eeguana_0.1.10.9001
#> 
#> loaded via a namespace (and not attached):
#>  [1] rstudioapi_0.14   knitr_1.42        magrittr_2.0.3    munsell_0.5.0    
#>  [5] tidyselect_1.2.0  colorspace_2.1-0  R6_2.5.1          rlang_1.1.3      
#>  [9] fastmap_1.1.1     fansi_1.0.6       tools_4.1.2       grid_4.1.2       
#> [13] tidytable_0.11.0  gtable_0.3.4      data.table_1.15.2 xfun_0.42        
#> [17] utf8_1.2.4        cli_3.6.2         withr_3.0.0       htmltools_0.5.4  
#> [21] yaml_2.3.7        digest_0.6.31     tibble_3.2.1      lifecycle_1.0.4  
#> [25] purrr_1.0.2       ggplot2_3.5.0     vctrs_0.6.5       fs_1.6.0         
#> [29] glue_1.7.0        evaluate_0.20     rmarkdown_2.20    reprex_2.1.0     
#> [33] compiler_4.1.2    pillar_1.9.0      scales_1.3.0      generics_0.1.3   
#> [37] pkgconfig_2.0.3
```

<sup>Created on 2024-03-11 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

bnicenboim, 2024-03-12:
thanks, this is fixed now.

Feel free to use https://github.com/bnicenboim/eeguana/discussions  if you have general questions

closed 2024-03-12 by bnicenboim