bnicenboim/eeguana #239: Package installation fails, with error:  Line starting 'VignetteB ...' is malformed!

kokoelma, 2024-04-24:
*Brief description of the problem*
Thanks for the eeguana package. When trying to install the package, the installation fails with the following error:

```r
> devtools::install_github("bnicenboim/eeguana")
Downloading GitHub repo bnicenboim/eeguana@HEAD
Error: Failed to install 'eeguana' from GitHub:
  Line starting 'VignetteB ...' is malformed!
```

*Operative system:*

Ubuntu 22.04.4 LTS

*Output of `sessionInfo()`:*

```
R version 4.3.3 (2024-02-29)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 22.04.4 LTS

Matrix products: default
BLAS:   /usr/lib/x86_64-linux-gnu/openblas-pthread/libblas.so.3 
LAPACK: /usr/lib/x86_64-linux-gnu/openblas-pthread/libopenblasp-r0.3.20.so;  LAPACK version 3.10.0

locale:
 [1] LC_CTYPE=C           LC_NUMERIC=C         LC_TIME=C            LC_COLLATE=C         LC_MONETARY=C        LC_MESSAGES=C        LC_PAPER=de_DE.UTF-8
 [8] LC_NAME=C            LC_ADDRESS=C         LC_TELEPHONE=C       LC_MEASUREMENT=C     LC_IDENTIFICATION=C 

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

loaded via a namespace (and not attached):
 [1] htmlwidgets_1.6.4   devtools_2.4.5      remotes_2.5.0       lattice_0.22-6      tzdb_0.4.0          numDeriv_2016.8-1.1 vctrs_0.6.5         tools_4.3.3        
 [9] curl_5.2.1          stats4_4.3.3        cAIC4_1.0           parallel_4.3.3      tibble_3.2.1        fansi_1.0.6         ucminf_1.2.1        pkgconfig_2.0.3    
[17] Matrix_1.6-5        lifecycle_1.0.4     compiler_4.3.3      stringr_1.5.1       carData_3.0-5       httpuv_1.6.15       htmltools_0.5.8.1   usethis_2.2.3      
[25] later_1.3.2         pillar_1.9.0        car_3.1-2           nloptr_2.0.3        urlchecker_1.0.1    MASS_7.3-60.0.1     ellipsis_0.3.2      cachem_1.0.8       
[33] sessioninfo_1.2.2   boot_1.3-30         abind_1.4-5         nlme_3.1-164        mime_0.12           digest_0.6.35       mvtnorm_1.2-4       stringi_1.8.3      
[41] purrr_1.0.2         forcats_1.0.0       splines_4.3.3       ordinal_2023.12-4   fastmap_1.1.1       grid_4.3.3          RLRsim_3.1-8        cli_3.6.2          
[49] magrittr_2.0.3      pkgbuild_1.4.4      utf8_1.2.4          readr_2.1.5         promises_1.3.0      lme4_1.1-35.2       hms_1.1.3           memoise_2.0.1      
[57] shiny_1.8.1.1       haven_2.5.4         miniUI_0.1.1.1      mgcv_1.9-1          profvis_0.3.8       rlang_1.1.3         Rcpp_1.0.12         xtable_1.8-4       
[65] glue_1.7.0          pkgload_1.3.4       rstudioapi_0.16.0   minqa_1.2.6         R6_2.5.1            fs_1.6.3      
```
Thank you.

bnicenboim, 2024-04-24:
it seems to be a problem on your side, there are several people having this issue with different packages:

https://duckduckgo.com/?q=Downloading+GitHub+repo+Error%3A+Failed+to+install+from+GitHub%3A+Line+starting+%27VignetteB+...%27+is+malformed!&t=brave&ia=web

Things to try:

1.  re-installing devtools from CRAN in a fresh R session.
```r
install.packages("devtools")
```

2. run the following:
```
Sys.setenv(LANG = “en_US.UTF-8”)
remotes::install_github("bnicenboim/eeguana")
```

kokoelma, 2024-04-24:
Thanks for the suggestions.  I installed devtools in a fresh R session, and tried installing eeguana, but the error still remained. And the result of option 2 was also not different (with Sys.setenv(LANG = "en_US").

However, following https://stackoverflow.com/a/24782341, I could manually download and install the package after installing its dependencies. Here's a log extract for reference.

```r
> library(downloader)
> download("https://github.com/bnicenboim/eeguana/archive/master.tar.gz", "eeguana.tar.gz")
...
> install.packages("eeguana.tar.gz", repos = NULL, type = "source")
...
ERROR: dependencies 'MBA', 'edfReader', 'fICA', 'JADE', 'RcppRoll', 'fastICA', 'tidytable', 'gsignal' are not available for package 'eeguana'

> pkgs <- c('MBA', 'edfReader', 'fICA', 'JADE', 'RcppRoll', 'fastICA', 'tidytable','gsignal') 
> install.packages(pkgs)

> install.packages("eeguana.tar.gz", repos = NULL, type = "source")
...
* DONE (eeguana)
> library(eeguana)
eeguana version 0.1.11.9001
An introduction to the package can be found in https://bruno.nicenboim.me/eeguana/articles/intro.html

Attaching package: 'eeguana'

The following objects are masked from 'package:eegUtils':

    channel_names, eeg_downsample, eeg_filter, events, events<-
>
```
Thanks again.